MagicMirrorOrg/MagicMirror #3761: [Feature Request] Allow to make an display order of module position in config

bugsounet, 2025-04-05:
### Prerequisites

- [x] I am running the latest version of MagicMirror², and know that this feature is not available now.
- [x] I know my issue is not related to a third-party module.
- [x] I have searched for [existing issues](https://github.com/MagicMirrorOrg/MagicMirror/issues) that already include this feature request, without success.

### Describe the Feature Request

Actually, we can't define order display of modules when using the same position

We have copy and past module config content to move it

I suggest to use a new feature on config to do this

### Describe the Use Case

```js
   {
      disabled: false,
      module: "clock",
      position: "top_left"
   },
   {
      disabled: false,
      module: "calendar",
      position: "top_left",
      config: {
        <--- calandar config--->
      }
   },
```

Sample: I want to display `calendar` before `clock` without copy/past any module config

### Describe Preferred Solution

Add a new feature named `order`

This feature allows to change the display order

```js  
   {
      disabled: false,
      module: "clock",
      position: "top_left",
      order: 2
   },
   {
      disabled: false,
      module: "calendar",
      position: "top_left",
      order: 1
      config: {
        <--- calandar config--->
      }
   },
```

in this case: display `calendar`  and below `clock`

### Participation

- [x] I am willing to submit a pull request for this change.

sdetweil, 2025-04-05:
that feature and property are already part of MMM-Config.

bugsounet, 2025-04-05:
@sdetweil:
 * Thanks to code your graphic part.
 * Maybe learn nodejs, is the best way for you ?
 * I don't care about your shitty module

rejas, 2025-04-05:
Please watch your language @bugsounet

bugsounet, 2025-04-05:
So, if you want the posts to go well: just tell to your collaborator (that i hate) ---≥ not to react to my posts

✓ everything will be fine ;)

ChrisF1976, 2025-04-13:
You made a dramatic exit, declared your work as ‘personal use only,’ locked your repos down – and now you’re back with a pull request like nothing happened? Sorry, but you can’t have it both ways. Either you’re part of the community, with all the give and take that comes with it – or you’re not. Might be time for some serious self-reflection.

bugsounet, 2025-05-08:
Please do not mix MM² and third-party modules.
They are two different things.

We're not going to start the debate here.

bugsounet, 2025-05-08:
@rejas or @KristjanESPERANTO  : can you lock this issue

Thx