MagicMirrorOrg/MagicMirror #3765: [Bug] {{ brief description }} Mac OS strange behavior compared to Docker

Rene1695, 2025-04-20:
### Environment

MagicMirror² version: v2.31.0
Node version: v22.14.0
npm version: v10.9.2
Platform: Mac OS

### Which start option are you using?

npm run start

### Are you using PM2?

No

### Module

None

### Have you tried disabling other modules?

- [x] Yes
- [ ] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

<details>
<summary>Configuration</summary>

```
<!-- Paste your configuration here. Don't forget to remove any sensitive information! -->
```
</details>

```js
<!-- Paste relevant code here -->
```

Steps to reproduce the issue:

### What did you expect to happen?

Same output as in docker environment

### What actually happened?

Some attributes from the config.js will be ignored if I use the installation on Mac OS and Electron. 

### Additional comments

I have a Mac OS arm server running in my home lab. I run MagicMirror in Docker quit a long time and it is working absolutely fine in Docker with my configuration. The only problem was that my safari browser needed to refresh the page from time to time. I found out that my setup is not the standard way of doing it so ChatGPT mentioned to install it directly on Mac and using Electron

So I installed it directly via npm on Mac OS and shifted all the config and css files and of course installed all my modules. I used GrafanaChart, PublicTransportHafas, OpenWeatherForecast and some standard ones. Once I start the server, some modules failed to load data. After investigating it I found out that some of the options I have set in the config.js will be ignored. But the general shape and all that stuff is working fine. It seems that only some attributes of the config.js will be ignored and I don't know why. 

In Docker the setup is working completely fine. On Mac directly not. Someone may also done it on Mac OS and have some tips for me? I have no idea were to look and also havent worked with electron yet. Are there major differences between Docker and Electron that I need to consider? 

### Participation

- [ ] I am willing to submit a pull request for this change.

sdetweil, 2025-04-20:
I run  MM on macos arm, without problems.. 
also on docker (intel) on my synology NAS

I don't see any differences

can you tell me about 'properties' not used/ignored...

sdetweil, 2025-04-20:
you shouldn't have to think about electron, it jsut happens to be the browser we open the page with .

sdetweil, 2025-04-20:
if you open the page with safari or chrome it should be the same as electron

sdetweil, 2025-04-20:
dont believe chatgpt..

Rene1695, 2025-04-20:
> I run  MM on macos arm, without problems.. 
> also on docker (intel) on my synology NAS
> 
> I don't see any differences
> 
> can you tell me about 'properties' not used/ignored...

Thanks for replying so quickly!

For example in MM-OpenWeatherForecast I set the BaseURL to the new 3.0 one call api. The default API is 2.5 an outdated. Although it is set so 3.0, the module uses the 2.5 default API. I checked that by adding logs in the node_helper.js

Same with the PublicTransportHafas. I set a different profile since the default profile "db" is outdated. The module still uses the default db profile. 

All other properties from the config.js will be paresed correctly. For example my location is set correctly I can see that in the logs on startup. 

I don't know in docker all of this works as expected. I don't understand why some properties will parsed correctly and some others will be ignored and set back to default.

Should I just stick with the docker build and instead of using safari just open it in electron or would it be even better to directly install on Mac OS? By the way: What is the difference between "npm start" and "npm run start"?

sdetweil, 2025-04-20:
>For example in MM-OpenWeatherForecast I set the BaseURL to the new 3.0 one call api. The default API is 2.5 an outdated.

the default baseURL is for NO particular version, BUT the default for the last 2 release is apiVersion:"3.0", so you should  NOT set 
ANY of the base url, apiVersion, endpoint variables as it uses the /onecall api per the openweather doc

>PublicTransportHafas. I set a different profile since the default profile "db" is outdated. The module still uses the default db profile.

you would have to show me,   remember MM is case SENSITIVE ON ALL PLATFORMS.. so test is NOT the same as Test
NO module checks for parms it does not expect, it they just ignore them

either implementation is fine,  again it should be identical 
(ps, did you install MM on Mac with the (my) script, or manually)?

npm takes commands as 
npm run xxxx (as listed in the scripts section of the package.json) 
'most' of the time its npm run start , BUT people said they hated typing the extra 'run', so npm start is a shortcut builtin
functionally no different

sdetweil, 2025-04-20:
soory, I don't know what MM-Openweaterforecast module this is .. 
I was speaking of the default weather module..

can you do 
git remote -v 
in the module folder so I can look at where it came from

sdetweil, 2025-04-20:
because you are using server mode in docker, there is no defined browser..
you can use any browser anywhere that can reach the docker host.. phone, laptop, ...  with any browser you like..

I don't use safari very often

Rene1695, 2025-04-20:
Don't worry. Here is the link to the module 

https://github.com/jclarke0000/MMM-OpenWeatherForecast

Also this is the other one I mentioned:

https://github.com/raywo/MMM-PublicTransportHafas

I took the config and css files from docker and put it in the directory's on Mac. So there should be no issues regarding case sensitivity or something because I didn't changed it.

I pulled the repo and went in the folder and did a "npm run install-mm" as mentioned on the website. Is this the script you mentioned?

Rene1695, 2025-04-20:
It says the installations process on RPi will take very long (like 10-20 minutes). In my case it was like 2 seconds. 

I know M1 is super fast but this seemed to quick in my opinion ...

sdetweil, 2025-04-20:
for openweatherforecast, that one is 4 years since last update
use the newer more active one
https://github.com/Tom-Hirschberger/MMM-OpenWeatherForecast

no change in parms. 

script, no

listed in the  alternative install  options,  (docker is one) , the script is listed first for install and upgrade

I don't see any  database config listed as option in the [MMM-PublicTransportHafas](https://github.com/raywo/MMM-PublicTransportHafas) module

sdetweil, 2025-04-20:
I know M1 is super fast but this seemed to quick in my opinion ...

M1 DISK is WAY faster than SD card too..

sdetweil, 2025-04-20:
also, this PublicTransport module as last update is 7 years ago,,

I'd use this new fork 
https://github.com/KristjanESPERANTO/MMM-PublicTransportHafas

sdetweil, 2025-04-20:
I see 'database' is hidden under profile

these are the listed profile choices..  https://github.com/public-transport/hafas-client/blob/main/p/readme.md

nothing says that are or are not supported on different  OS/hardware systems

Rene1695, 2025-04-20:
You got it! I was using this modules in the past and now I pulled the wrong repos... sorry for being dump and thank your for the help :)

closed 2025-04-20 by Rene1695

sdetweil, 2025-04-20:
now that you have happy running system, you should insure you have a backup 

see my https://github.com/sdetweil/MagicMirror-backup-restore scripts,
and install/upgrade are here  
https://github.com/sdetweil/MagicMirror_scripts

Rene1695, 2025-04-20:
awesome! thank you so so much :)