MagicMirrorOrg/MagicMirror #3785: [Bug] {{ fatal syntax error in config.js not detected, but MM dies }}

sdetweil, 2025-05-11:
### Environment

doesn't matter

### Which start option are you using?

npm run start

### Are you using PM2?

No

### Module

None

### Have you tried disabling other modules?

- [x] Yes
- [x] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

<details>
<summary>Configuration</summary>

use the config.js.sample
add _ before let config
```
_
let config = {
```
</details>

Steps to reproduce the issue:

issue npm start 

### What did you expect to happen?

mm would report an error
npm run config:check reports no error 

### What actually happened?

```text
(base) sam@sams:~/MagicMirror$ npm start 

> magicmirror@2.32.0-develop start
> node --run start:x11

[2025-05-11 17:50:42.970] [LOG]   Starting MagicMirror: v2.32.0-develop 
[2025-05-11 17:50:42.984] [LOG]   Loading config ... 
[2025-05-11 17:50:42.985] [LOG]   config template file not exists, no envsubst 
[2025-05-11 17:50:43.085] [INFO]  Checking config file /home/sam/MagicMirror/config/config.js ... 
[2025-05-11 17:50:43.101] [INFO]  Your configuration file doesn't contain syntax errors :) 
[2025-05-11 17:50:43.101] [INFO]  Checking modules structure configuration ... 
[2025-05-11 17:50:43.120] [ERROR] _ is not defined 
```

mm ends immediately 

### Participation

- [x] I am willing to submit a pull request for this change.

KristjanESPERANTO, 2025-05-13:
Not about this issue: When I touch the check boxes when I read the issue on my smartphone the checkbox toggles get saved. That's not nice. 

About that issue: I'm on a business trip but will be home soon. I can have a look at it this week.

sdetweil, 2025-05-13:
yes, the validator only checks the modules:[] array

also doesnt catch _ between module entries

i have code that could catch them if the schema approach doesn't work

sdetweil, 2025-05-14:
here is my code for catching syntax errors not caught by schema checks, in js/check_config.js
```js
	// Scan all modules (line 104)
	const validate = ajv.compile(schema);
	let data=null
        // wrap require with try catch, to get the stacktrace of any errors discovered
	try {
	        data = require(configFileName);
	}
	catch(error){
		// get error and top line of stack trace
		const error_text=error.stack.split('\n').slice(0,2)

		// get the error text from the line
		const real_error= error_text[0].split(':')[1] // remove general error def ("Reference error")

		// get the file name, line number and column number from the stack trace data
                // get the stuff inside the parens, name , line and column numbers
		const fn=error_text[1].split('(')[1].split(')')[0].split(':')

		console.error(`fatal error in ${fn[0]} at line ${fn[1]} column ${fn[2]}, ${real_error}`)
		process.exit(1)
	}
        //  this depends on require succeeding, config.js is not json, so can't read in and parse
	const valid = validate(data);

KristjanESPERANTO, 2025-05-14:
I can confirm that your code works, but activating the rule `no-undef` seems to have the same effect with less changes. I created #3786 to show my approach. What do you think?

sdetweil, 2025-05-14:
All good.. didn't know if there was a rule you could apply

sdetweil, 2025-05-16:
@KristjanESPERANTO 

is there a way to catch this as well..

```
    ipWhitelist: [],
   a:b,   <---  b is undef
   modules:[
     { module:"a"
     },
    _    <--- _ is undef, and in the wrong place
    { module:"b"
    }
```

reports Parsing error: Unexpected token { 

but misses the undef of b, and _

KristjanESPERANTO, 2025-05-16:
I don't think it's possible to catch both in one step.  If there is a syntax error, the linter is unable to perform static analysis because the code cannot be parsed correctly. The user has to fix the syntax error first.

sdetweil, 2025-05-16:
funny,  search yields 
```text
To find undefined variables and syntax errors in a JavaScript file, a linter, such as ESLint, can be used. Linters analyze code statically, identifying potential issues without executing the code.
Installation and Setup (ESLint Example)
Install ESLint.
Code

    npm install -g eslint
Configure ESLint.
Code

     eslint --init
This command guides you through creating a configuration file (.eslintrc.js or similar) where you can specify rules. Enable no-undef rule.
In your ESLint configuration file, ensure the no-undef rule is enabled (set to "error" or "warn"). This rule flags the use of undeclared variables.
JavaScript

    // .eslintrc.js
    module.exports = {
      // ...
      rules: {
        "no-undef": "error",
        // ... other rules
      },
    };
Syntax error detection.
ESLint, by default, also catches syntax errors. No specific rule configuration is needed for basic syntax error detection.
Usage
Lint the file.
Code

    eslint your_javascript_file.js
Review the output.
ESLint will output any detected errors or warnings, including undefined variables and syntax errors, along with their locations in the file.
Example
JavaScript

// Example with an undefined variable and a syntax error
function example() {
  let x = 10;
  console.log(y); // Undefined variable: y
  if (x > 5 { // Syntax error: missing closing parenthesis
    console.log("x is greater than 5");
  }
}
example();
Linting this file with ESLint would produce output similar to:
Code

your_javascript_file.js
  line 3, col 13, Error - 'y' is not defined. (no-undef)
  line 4, col 11, Error - Expected ')' and instead saw '{'. (syntax)
This output indicates an undefined variable y on line 3 and a syntax error on line 4 due to a missing closing parenthesis.
```

KristjanESPERANTO, 2025-05-16:
That's output from a chatbot, right?

sdetweil, 2025-05-16:
yes

KristjanESPERANTO, 2025-05-16:
I am quite sure that the chatbot is wrong. But feel free to prove me wrong 🙂 

By the way: If you quote a chatbot, please indicate this. This makes it easier to understand such statements.

sdetweil, 2025-05-16:
understood, didn't think of it til you asked.. will in the future

closed 2025-06-30 by khassel