MagicMirrorOrg/MagicMirror #3790: [Bug] Start issue with electron v36 on GNOME

KristjanESPERANTO, 2025-05-23:
### Environment

MagicMirror² version: `v2.32.0-develop`

```bash
[2025-05-24 00:23:50.812] [INFO]  System information:
### SYSTEM:   manufacturer: Notebook; model: N650DU; virtual: false
### OS:       platform: linux; distro: Debian GNU/Linux; release: 12; arch: x64; kernel: 5.10.0-20-amd64
### VERSIONS: electron: 36.3.0; used node: 24.1.0; installed node: 24.1.0; npm: 10.9.0; pm2: 6.0.6
### OTHER:    timeZone: Europe/Berlin; ELECTRON_ENABLE_GPU: undefined 
```

### Which start option are you using?

npm run start:wayland

### Are you using PM2?

No

### Module

None

### Have you tried disabling other modules?

- [x] Yes
- [ ] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

<details>
<summary>Configuration</summary>

```js
let config = 
{
	logLevel: ["INFO", "LOG", "WARN", "ERROR", "DEBUG"], // Add "DEBUG" for even more logging
  address: '0.0.0.0',
  ipWhitelist: [],
  modules: [
    {
      module: 'clock',
      position: 'top_left',
      config: {}
    },
  ],
  useHttps: false,
  httpsPrivateKey: '',
  httpsCertificate: ''
}

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}
```
</details>

Steps to reproduce the issue:

- running updated develop branch on a GNOME system
- start `node --run start:wayland`, `node --run start`

### What did you expect to happen?

MM starts

### What actually happened?

MM doesn't start:

```bash
kristjan@debian:~/MagicMirror$ node --run start:wayland
[2025-05-24 00:36:55.163] [LOG]   Starting MagicMirror: v2.32.0-develop 
[2025-05-24 00:36:55.180] [LOG]   Loading config ... 
[2025-05-24 00:36:55.182] [LOG]   config template file not exists, no envsubst 
[2025-05-24 00:36:55.444] [INFO]  Checking config file /home/kristjan/MagicMirror/config/config.js ... 
[2025-05-24 00:36:55.484] [INFO]  Your configuration file doesn't contain syntax errors :) 
[2025-05-24 00:36:55.485] [INFO]  Checking modules structure configuration ... 
[2025-05-24 00:36:55.532] [INFO]  Your modules structure configuration doesn't contain errors :) 
[2025-05-24 00:36:55.533] [LOG]   Loading module helpers ... 
[2025-05-24 00:36:55.534] [LOG]   No helper found for module: clock. 
[2025-05-24 00:36:55.534] [LOG]   All module helpers loaded. 
[2025-05-24 00:36:55.541] [LOG]   Starting server on port 8080 ...  
[2025-05-24 00:36:55.542] [WARN]  You're using a full whitelist configuration to allow for all IPs 
[2025-05-24 00:36:55.891] [LOG]   Server started ... 
[2025-05-24 00:36:55.891] [LOG]   Sockets connected & modules started ... 

(process:108168): Gtk-ERROR **: 00:36:55.906: GTK 2/3 symbols detected. Using GTK 2/3 and GTK 4 in the same process is not supported
/home/kristjan/MagicMirror/node_modules/electron/dist/electron exited with signal SIGTRAP
```

### Additional comments

That is a known issue for Electron 36: https://www.electronjs.org/blog/electron-36-0#behavior-changed-gtk-4-is-the-default-when-running-on-gnome

**Workaround**

Append ` -- --gtk-version=3` to your start command. Like this:

- `npm run start -- --gtk-version=3`
- `npm run start:wayland -- --gtk-version=3`
- `node --run start -- --gtk-version=3`
- `node --run start:wayland -- --gtk-version=3`

Server mode is not affected.

Sooner or later, someone else might stumble across this too, so I wanted to make a note of it here. Maybe someone has a better solution than the workaround.

### Participation

- [ ] I am willing to submit a pull request for this change.

sdetweil, 2025-05-23:
can you dump the env before starting MagicMirror to
see if there is any ident info we might use under the covers if we have to

env

KristjanESPERANTO, 2025-05-23:
Sure: 

```bash
kristjan@debian:~/MagicMirror$ env
SHELL=/bin/bash
SESSION_MANAGER=local/debian:@/tmp/.ICE-unix/2378,unix/debian:/tmp/.ICE-unix/2378
QT_ACCESSIBILITY=1
COLORTERM=truecolor
SSH_AGENT_LAUNCHER=openssh
NVM_INC=/home/kristjan/.nvm/versions/node/v24.1.0/include/node
XDG_MENU_PREFIX=gnome-
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
GNOME_KEYRING_CONTROL=/run/user/1000/keyring
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
XMODIFIERS=@im=ibus
DESKTOP_SESSION=gnome
GTK_MODULES=gail:atk-bridge
PWD=/home/kristjan/MagicMirror
LOGNAME=kristjan
XDG_SESSION_DESKTOP=gnome
XDG_SESSION_TYPE=wayland
PNPM_HOME=/home/kristjan/.local/share/pnpm
SYSTEMD_EXEC_PID=2754
XAUTHORITY=/run/user/1000/.mutter-Xwaylandauth.GAE862
GDM_LANG=eo.UTF-8
HOME=/home/kristjan
USERNAME=kristjan
IM_CONFIG_PHASE=1
LANG=eo.UTF-8
XDG_CURRENT_DESKTOP=GNOME
VTE_VERSION=7006
WAYLAND_DISPLAY=wayland-0
GNOME_TERMINAL_SCREEN=/org/gnome/Terminal/screen/bf8281fb_abf3_45f1_ad4a_5c0823b41c23
NVM_DIR=/home/kristjan/.nvm
GNOME_SETUP_DISPLAY=:1
XDG_SESSION_CLASS=user
TERM=xterm-256color
USER=kristjan
GNOME_TERMINAL_SERVICE=:1.215
DISPLAY=:0
SHLVL=1
NVM_CD_FLAGS=
MOZ_ENABLE_WAYLAND=1
QT_IM_MODULE=ibus
XDG_RUNTIME_DIR=/run/user/1000
BUN_INSTALL=/home/kristjan/.bun
XDG_DATA_DIRS=/usr/share/gnome:/home/kristjan/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop
PATH=/home/kristjan/.local/share/pnpm:/home/kristjan/.bun/bin:/home/kristjan/.nvm/versions/node/v24.1.0/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/snap/bin:/home/kristjan/.local/bin
GDMSESSION=gnome
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
NVM_BIN=/home/kristjan/.nvm/versions/node/v24.1.0/bin
OLDPWD=/home/kristjan
_=/usr/bin/env
```

KristjanESPERANTO, 2025-07-27:
The problem doesn't occur with electron 37 in the develop branch any more 😃