MagicMirrorOrg/MagicMirror #3796: [Bug] Nodejs v24 lacks binaries for arm32

khassel, 2025-05-31:
### Environment

MagicMirror² version: irrelevant
Node version: v24.x
npm version: irrelevant
Platform: arm32

### Which start option are you using?

npm run start

### Are you using PM2?

No

### Module

None

### Have you tried disabling other modules?

- [x] Yes
- [ ] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

Trying to install nodejs v24 on an arm32 device

### What did you expect to happen?

Finding binaries under https://nodejs.org/dist/latest-v24.x/ and docker images for arm32.

### What actually happened?

The support for arm32 was declared as experimental in nodejs v24. Concerning [this comment](https://github.com/nodejs/docker-node/issues/2232#issuecomment-2867510141) compiling of nodejs v24 fails under arm32.

Nodejs v24 is the upcoming lts release. The current lts release v22 ends 30 Apr 2027.

As consequence we should not deprecate nodejs v24 in this projects as long as possible.

### Participation

- [ ] I am willing to submit a pull request for this change.

sdetweil, 2025-06-01:
yes, and only in 64 bit mode, which has a bigger working set.

we have already dropped pi0w

sdetweil, 2025-08-02:
Still following for next release

khassel, 2025-11-07:
Nodejs v24 is meanwhile the current lts version.

sdetweil, 2025-11-07:
and still without armv7l binaries

khassel, 2025-11-08:
> and still without armv7l binaries

I think they will not come back ... so we are running into this problem faster than expected because upcoming electron v40 already uses node v24.

khassel, 2025-11-08:
yes, to avoid running an unsupported electron version we would have to migrate to electron v40 (and with that to nodejs v24) with the April release

sdetweil, 2025-11-08:
yeh, dang

khassel, 2025-11-08:
one solution could be to patch back the `package.json` to an electron v39 version (and node engine stuff) so arm32 users can still use newer mm versions on their own risk. We could provide such e.g. `sed` statements in the docs and @sdetweil could do this in his scripts.

My first idea to this, maybe we find better solutions until April ...

khassel, 2026-09-02:
Next problem with arm32: After updating electron to v44 my container builds for arm32 failed with:

```bash
21.70 Error: Electron v44.1.1 does not have a published linux/armv7l artifact. Electron 44 and newer no longer publish win32/ia32 or linux/armv7l builds; the last release line with 32-bit support is Electron 43, which is supported until January 2027.
```

Excerpt from https://www.electronjs.org/docs/latest/breaking-changes:

> Removed: Windows 32-bit (ia32) and Linux 32-bit ARM (armv7l) support

> Electron no longer publishes prebuilt binaries for 32-bit platforms: Windows x86 (win32-ia32) and Linux ARM (linux-armv7l). All related release artifacts (chromedriver, mksnapshot, ffmpeg, and the Windows x86 node.lib on the Electron headers CDN) are no longer published either.

> Older versions of Electron will continue to support these platforms, but Electron v44.0.0 and higher will only be published for 64-bit platforms.

> Once the v43 series reaches end of life in January 2027, these 32-bit platforms will no longer be supported.

We can, of course, stick with Electron v43 until the end of the year, but that only buys us time. We will have to upgrade to v44 by the January release at the latest.

sdetweil, 2026-09-02:
I can start the conversation on the forum.. I'd like to hold  electron 43 til 1/1 release, give users time to move..(or decision to stay behind for  a while) , cause many have the smaller hardware 02w, and pi3..

khassel, 2026-09-02:
the error above is raised when the electron binaries are downloaded, the `node --run install-mm` works.

One simple workaround is to run `npm install electron@43` after the above command.

sdetweil, 2026-09-02:
Ok, decent workaround 

I can fix installer and upgrade too

sdetweil, 2026-09-03:
See
https://forum.magicmirror.builders/post/131855

KristjanESPERANTO, 2026-09-12:
So, am I understanding this correctly: We can close this issue with the next release, since we will no longer officially support arm32 and a workaround is documented in the forum. Right?

khassel, 2026-09-12:
I would leave the issue open for the sake of visibility and further discussion.

The workaround using Electron v43 is only temporary; from January onwards, using this solution will be "at your own risk" since there will be no further security updates.

One option for 32-bit systems would be a different browser that displays an MM-server-only instance. But perhaps better solutions can be found.

sdetweil, 2026-09-12:
I’d also Leave it open 
I do other browser over server already in install

KristjanESPERANTO, 2026-09-12:
Okay, I get your points :+1: