MagicMirrorOrg/MagicMirror #3797: [Bug] Calendar shows wrong time for recurring events created before DST

plebcity, 2025-06-01:
### Environment

MagicMirror² version: 2.31.0
Node version: 22.16.0
npm version: 10.9.2
Platform: Debian Linux Docker container (karsten13/magicmirror:latest)

### Which start option are you using?

npm run server

### Are you using PM2?

No

### Module

calendar

### Have you tried disabling other modules?

- [ ] Yes
- [x] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

I'm loading a .ics calendar with all default config except maximumNumberOfDays (2) and fetchInterval of 15 minutes.

When I create a recurring event in March, before daylight saving time then after daylight saving time it will show 1 hour too late (10 instead of 9 for example)

### What did you expect to happen?

I expected the calendar module to work with recurring events combined with DST

### What actually happened?

It shows 1 hour too late for every recurring event that is created in the winter period (before DST). 

### Additional comments

The problem lies in the interpretation of the results from RRule, as per the logging below (I have seperated them to place comments on relevant lines):

```
[2025-05-31 11:45:54.434] [DEBUG] start: Tue Mar 11 2025 09:00:00 GMT+0100 (Central European Standard Time) 
[2025-05-31 11:45:54.434] [DEBUG] end:: Tue Mar 11 2025 09:15:00 GMT+0100 (Central European Standard Time) 
```
This seems to be interpreted fine, the event first occurence was on tuesday the 11th of March 2025 at 09:00 in the morning at Europe/Brussels time. So I would expect to see an event on monday the 2nd of May 2025 at 09:00 as this only occurs on workdays.

```
[2025-05-31 11:45:54.434] [DEBUG] duration: 900000 
[2025-05-31 11:45:54.434] [DEBUG] title: xxxxxxxxxxxxxxxxxxx
[2025-05-31 11:45:54.435] [DEBUG] Search for recurring events between: Sat May 31 2025 11:45:53 GMT+0200 (Central European Summer Time) and Mon Jun 02 2025 23:59:59 GMT+0200 (Central European Summer Time) 
```
This shows that the calendar module searching for the correct range of recurring events. Below we can see how that was passed to RRule.

```
[2025-05-31 11:45:54.435] [DEBUG] RRule: DTSTART;TZID=Europe/Brussels:20250311T090000
RRULE:FREQ=WEEKLY;BYDAY=FR,MO,TH,TU,WE 
[2025-05-31 11:45:54.435] [DEBUG] Title: xxxxxxxxxxxxxxxx, with dates: ["2025-06-02T09:00:00.000Z","2025-06-03T09:00:00.000Z"] 
```
RRule got the original DTStart from the .ics file correct, as you can see there is a timezone Europe/Brussels that is not part of the ISO-8601 datetime after it. But on this line: https://github.com/MagicMirrorOrg/MagicMirror/blob/master/modules/default/calendar/calendarfetcherutils.js#L302 the timezone is removed from the options passed to RRule which means RRule assumes it is in UTC, see: https://github.com/jkbrzt/rrule?tab=readme-ov-file#important-use-utc-dates

```
[2025-05-31 11:45:54.435] [DEBUG] Rule has byweekday, checking for correction 
[2025-05-31 11:45:54.435] [DEBUG] initial tz=Europe/Brussels 
[2025-05-31 11:45:54.435] [DEBUG] corrected tz=Europe/Brussels 
[2025-05-31 11:45:54.435] [DEBUG] start date/time=Tue Mar 11 2025 09:00:00 GMT+0100 (Central European Standard Time) 
[2025-05-31 11:45:54.435] [DEBUG] start offset=60 
[2025-05-31 11:45:54.435] [DEBUG] start date/time w tz =Tue Mar 11 2025 09:00:00 GMT+0100 (Central European Standard Time)
```
This seems fine but it is really weird that the calendar module tries to do anything with the original timezone etc.
```
[2025-05-31 11:45:54.435] [DEBUG] event date=Mon Jun 02 2025 11:00:00 GMT+0200 (Central European Summer Time) 
[2025-05-31 11:45:54.435] [DEBUG] event offset=120 hour=11 event date=Mon Jun 02 2025 11:00:00 GMT+0200 (Central European Summer Time) 
[2025-05-31 11:45:54.435] [DEBUG] offset 
[2025-05-31 11:45:54.435] [DEBUG] adjust down 1 hour dst change 
[2025-05-31 11:45:54.435] [DEBUG] adjustHours=-1 
[2025-05-31 11:45:54.435] [DEBUG] Applying timezone adjustment hours=-1 to Mon Jun 02 2025 11:00:00 GMT+0200 (Central European Summer Time) 
```
This is where it is broken, it has interpreted the result of RRule as if the datetime returned has the UTC timezone, therefor it has corrected the time to 11:00 and then for some reason there is a -1 hour adjustment???

What I think should happen is remove the Z from the returned datetime from RRule and then add the local timezone, not convert it to the local timezone. Don't ever work with just the offset if you are calculating with datetimes, always use the zone.

### Participation

- [ ] I am willing to submit a pull request for this change.

sdetweil, 2025-06-01:
Please provide the ical event that exhibits this issue

curl -sL calendar_Url >somefile.txt

find the vevent in  somefile.txt
edit xxx the personal info, dont remove any lines

plebcity, 2025-06-01:
```
BEGIN:VEVENT
DTSTART;TZID=Europe/Brussels:20250311T090000
DTEND;TZID=Europe/Brussels:20250311T091500
RRULE:FREQ=WEEKLY;BYDAY=FR,MO,TH,TU,WE
DTSTAMP:20250531T091103Z
ORGANIZER;CN=xxxxxxxxxxxxxxxx
UID:xxxxxxxxxxxxxxxxx
ATTENDEE;CUTYPE=INDIVIDUAL;ROLE=REQ-PARTICIPANT;PARTSTAT=ACCEPTED;CN=xxxxxxxxxxxxxxxxxxxxx;X-NUM-GUESTS=0:mailto:xxxxxxxxxxxxxxxxx
ATTENDEE;CUTYPE=INDIVIDUAL;ROLE=REQ-PARTICIPANT;PARTSTAT=ACCEPTED;CN=xxxxxxxxxxxxxxxxxxxxxxxx;X-NUM-GUESTS=0:mailto:xxxxxxxxxxxxxxxxxxxxxx
ATTENDEE;CUTYPE=INDIVIDUAL;ROLE=REQ-PARTICIPANT;PARTSTAT=ACCEPTED;CN=xxxxxxxxxxxxxxxxxx;X-NUM-GUESTS=0:mailto:xxxxxxxxxxxxxxxxxxxxxxxxx
ATTENDEE;CUTYPE=INDIVIDUAL;ROLE=REQ-PARTICIPANT;PARTSTAT=ACCEPTED;CN=xxxxxxxxxxxxxxxxxxxxxxxx;X-NUM-GUESTS=0:mailto:xxxxxxxxxxxxxxxxxxxxxx
ATTENDEE;CUTYPE=INDIVIDUAL;ROLE=REQ-PARTICIPANT;PARTSTAT=ACCEPTED;CN=xxxxxxxxxxxxxxxxxxxxx;X-NUM-GUESTS=0:mailto:xxxxxxxxxxxxxxxxx
ATTENDEE;CUTYPE=INDIVIDUAL;ROLE=REQ-PARTICIPANT;PARTSTAT=ACCEPTED;CN=xxxxxxxxxxxxxxxxxxxxxx;X-NUM-GUESTS=0:mailto:xxxxxxxxxxxxxxxxxxxxxx
ATTENDEE;CUTYPE=INDIVIDUAL;ROLE=REQ-PARTICIPANT;PARTSTAT=ACCEPTED;CN=xxxxxxxxxxxxxxxxxxxxxx;X-NUM-GUESTS=0:mailto:xxxxxxxxxxxxxxxxxxxxxxx
ATTENDEE;CUTYPE=INDIVIDUAL;ROLE=REQ-PARTICIPANT;PARTSTAT=ACCEPTED;CN=xxxxxxxxxxxxxxxxxxxxx;X-NUM-GUESTS=0:mailto:xxxxxxxxxxxxxxxxxxxxx
ATTENDEE;CUTYPE=INDIVIDUAL;ROLE=REQ-PARTICIPANT;PARTSTAT=NEEDS-ACTION;CN=xxxxxxxxxxxxxxxxxxxxx;X-NUM-GUESTS=0:mailto:xxxxxxxxxxxxxxxxxxxx
ATTENDEE;CUTYPE=INDIVIDUAL;ROLE=REQ-PARTICIPANT;PARTSTAT=ACCEPTED;CN=xxxxxxxxxxxxxxxxxxxxxx;X-NUM-GUESTS=0:mailto:xxxxxxxxxxxxxxxxxxxxxxxx
ATTENDEE;CUTYPE=INDIVIDUAL;ROLE=REQ-PARTICIPANT;PARTSTAT=ACCEPTED;CN=xxxxxxxxxxxxxxxxxxxxxxxxxx;X-NUM-GUESTS=0:mailto:xxxxxxxxxxxxxxxxxxxxxxxxxxxx
X-GOOGLE-CONFERENCE:xxxxxxxxxxxxxxxxxxxxxxxxxx
CREATED:20230111T114612Z
DESCRIPTION:xxxxxxxxxxxxxxxxxxxxxxxxxxx
LAST-MODIFIED:20250528T071312Z
SEQUENCE:1
STATUS:CONFIRMED
SUMMARY:xxxxxxxxxxxxxxxxxxxxxxx
TRANSP:OPAQUE
END:VEVENT
```
Here is the event from the .ics file, the relevant information from the event is also logged btw.

sdetweil, 2025-06-01:
RRULE cannot handle non-local dates, so the MagicMirror code converts to local date/time before rrule.between

and converts back to timezone based after

so if the date/time returned from rrule.between is incorrect, 
in the dates= list the final result will be wrong

i  am away from my system til tomorrow

plebcity, 2025-06-02:
> I made an ics from the above event and see this on my system set to Brussels timezone , 9am as expected

What if you change the start date to February? It might have something to do with DST in America for you as this event was created after DST in New York.

I'm currently trying to refactor the code in the filterEvents method to remove all of the different calculations and conversions between dates.

sdetweil, 2025-06-02:
@plebcity 
> I'm currently trying to refactor the code in the filterEvents method to remove all of the different calculations and conversions between dates.

good luck with that.  date/time/timezone/moment and RRULE all want different things.. RRULE is the biggest pain

plebcity, 2025-06-02:
> moved to Feb 11, and see the same results.. also adjusted to my Chicago tz and see 02:00 as expected
> 
> what tz is the docker container running?

Everything is running in Europe/Amsterdam (same rules as Brussels), docker container, host machine and client browser.

plebcity, 2025-06-02:
> good luck with that. date/time/timezone/moment and RRULE all want different things.. RRULE is the biggest pain

I'm trying to figure out what all these weird calculations do but I can't get my head around this method:

```
	fixEventtoLocal (event) {
		// if there are excluded dates, their date is incorrect and possibly key as well.
		if (event.exdate !== undefined) {
			Object.keys(event.exdate).forEach((dateKey) => {
				// get the date
				let exdate = event.exdate[dateKey];
				Log.debug("exdate w key=", exdate);
				//exdate=CalendarFetcherUtils.convertDateToLocalTime(exdate, event.end.tz)
				exdate = new Date(new Date(exdate.valueOf() - ((120 * 60 * 1000))).getTime());
				Log.debug("new exDate item=", exdate, " with old key=", dateKey);
				let newkey = exdate.toISOString().slice(0, 10);
				if (newkey !== dateKey) {
					Log.debug("new exDate item=", exdate, ` key=${newkey}`);
					event.exdate[newkey] = exdate;
					//delete event.exdate[dateKey]
				}
			});
			Log.debug("updated exdate list=", event.exdate);
		}
		if (event.recurrences) {
			Object.keys(event.recurrences).forEach((dateKey) => {
				let exdate = event.recurrences[dateKey];
				//exdate=new Date(new Date(exdate.valueOf()-(60*60*1000)).getTime())
				Log.debug("new recurrence item=", exdate, " with old key=", dateKey);
				exdate.start = CalendarFetcherUtils.convertDateToLocalTime(exdate.start, exdate.start.tz);
				exdate.end = CalendarFetcherUtils.convertDateToLocalTime(exdate.end, exdate.end.tz);
				Log.debug("adjusted recurringEvent start=", exdate.start, " end=", exdate.end);
			});
		}
		Log.debug("modified recurrences before rrule.between", event.recurrences);
	},
```

sdetweil, 2025-06-02:
the excluded and recurrence dates in the list also need to be changed to local for rrule

sdetweil, 2025-06-03:
there are testcases 
do
``` 
npm run test:electron
npm run test:e2e
```

plebcity, 2025-06-03:
> there are testcases do
> 
> ```
> npm run test:electron
> npm run test:e2e
> ```

I see some tests are failing, I will check those failures tomorrow, I see some exdates failures and some failures that probably have to do with the conversion from the unix timestamp back to a readable datetime string in the dom.