MagicMirrorOrg/MagicMirror #3835: undocumented breaking changes with 2.32.0 - express v5

skuethe, 2025-07-05:
### What problem do you want to solve with this change?

It seems you switched to [express version 5](https://github.com/MagicMirrorOrg/MagicMirror/pull/3774) with the latest update. Unfortunately there seems to be no hint to this in the release notes.
express v5 introduced breaking changes which are documented here:
https://expressjs.com/en/guide/migrating-5.html

This is especially critical for the `this.expressapp` helper you provide:
https://docs.magicmirror.builders/development/node-helper.html#this-expressapp

### What do you think is the correct solution?

It would be very helpful to add this hint as a breaking change (targeting module developers) in the release notes of v2.32.0

### Participation

- [ ] I am willing to submit a pull request for this change.

dathbe, 2025-07-05:
The breaking changes do not seem to be limited to upgrading the express version.  Here are additional apps that seem to refuse to work upon upgrade to 2.32.0:

updatenotification (will not load)
[MMM-CalendarExt2](https://github.com/MMM-CalendarExt2/MMM-CalendarExt2) (will not load)
[MMM-LogExt](https://github.com/MMRIZE/MMM-LogExt) (relies on express, though throws errors even when express v4 is explicitly loaded as a dependency)
[MMM-OpenWeatherMapForecast](https://github.com/MarcLandis/MMM-OpenWeatherMapForecast)
[MMM-MyStandings](https://github.com/dathbe/MMM-MyStandings)

sdetweil, 2025-07-05:
updatenotification loads for me on 2.32
calendarext2 is deprecated, use Ext3
MMM-LogExt works on 2.32

dathbe, 2025-07-05:
I'm going to delete my comment.  At least partially, my list is a symptom of not install node packages as we discussed last night.

sdetweil, 2025-07-05:
yes, when things het messed up, they get messed up..

delete node_modules folder and do another
npm install_mm

dathbe, 2025-07-05:
> calendarext2 is deprecated, use Ext3

"Only week and month views. I found that people are rarely interested in other views on CX2. So I drop out different views."
https://github.com/MMRIZE/MMM-CalendarExt3

calendarext2 is not deprecated, it just has a more updated version for _some_ functions.

khassel, 2025-07-05:
> I think I can update the release notes here and in the forum, will place a hint for this there later.

added a hint

skuethe, 2025-07-05:
Thank you <3 and keep up the good work :)

closed 2025-07-05 by skuethe

KristjanESPERANTO, 2025-07-06:
> Ext2 is no longer being developed or supported.

To set the records straight: Ext2 is maintained, not by by the original author but by another team. Ext3, however, is quasi unmaintained ([see the forum](https://forum.magicmirror.builders/topic/19672/long-time-no-see-and-need-a-help.?_=1751787062350)).

sdetweil, 2025-07-06:
Ok, wasn't aware it had shifted to a new team