MagicMirrorOrg/MagicMirror #3856: [Bug] Weather module handling of absent precipitationLastHour value

davidrea, 2025-08-07:
### Environment

MagicMirror² version: 2.32.0
Node version: 22.14.0
npm version: 10.9.2
Platform:

```
0|MagicMirror  | ### SYSTEM:   manufacturer: Raspberry Pi Foundation; model: Raspberry Pi 4 Model B Rev 1.5; virtual: false
0|MagicMirror  | ### OS:       platform: linux; distro: Debian GNU/Linux; release: 12; arch: arm64; kernel: 6.12.34+rpt-rpi-v8
0|MagicMirror  | ### VERSIONS: electron: 36.7.3; used node: 22.14.0; installed node: 22.14.0; npm: 10.9.2; pm2: 6.0.8
0|MagicMirror  | ### OTHER:    timeZone: America/New_York; ELECTRON_ENABLE_GPU: undefined
```

### Which start option are you using?

node --run start:x11

### Are you using PM2?

Yes

### Module

weather

### Have you tried disabling other modules?

- [x] Yes
- [ ] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

<details>
<summary>Configuration</summary>

```
{
        module: "weather",
        position: "top_center",
        config: {
                weatherProvider: "weathergov",
                type: "current",
                lat: 43,
                lon: -77,
                roundTemp: true
        }
},
```
</details>

The weather module's "current" display type fails with "Loading..." displayed on-screen when api.weather.gov returns station conditions with a missing key...

```js
currentWeather.precipitationAmount = currentWeatherData.precipitationLastHour.value ? currentWeatherData.precipitationLastHour.value : currentWeatherData.precipitationLast3Hours.value;
```

Steps to reproduce the issue:

Somehow get the weather.gov API to return an object (example follows) whose "properties" object does not contain a "precipitationLastHour" key...

<details>
<summary>currentWeatherData</summary>

```
{
    "@context": [
        "https://geojson.org/geojson-ld/geojson-context.jsonld",
        {
            "@version": "1.1",
            "wx": "https://api.weather.gov/ontology#",
            "s": "https://schema.org/",
            "geo": "http://www.opengis.net/ont/geosparql#",
            "unit": "http://codes.wmo.int/common/unit/",
            "@vocab": "https://api.weather.gov/ontology#",
            "geometry": {
                "@id": "s:GeoCoordinates",
                "@type": "geo:wktLiteral"
            },
            "city": "s:addressLocality",
            "state": "s:addressRegion",
            "distance": {
                "@id": "s:Distance",
                "@type": "s:QuantitativeValue"
            },
            "bearing": {
                "@type": "s:QuantitativeValue"
            },
            "value": {
                "@id": "s:value"
            },
            "unitCode": {
                "@id": "s:unitCode",
                "@type": "@id"
            },
            "forecastOffice": {
                "@type": "@id"
            },
            "forecastGridData": {
                "@type": "@id"
            },
            "publicZone": {
                "@type": "@id"
            },
            "county": {
                "@type": "@id"
            }
        }
    ],
    "id": "https://api.weather.gov/stations/KROC/observations/2025-08-07T02:40:00+00:00",
    "type": "Feature",
    "geometry": {
        "type": "Point",
        "coordinates": [
            -77.67,
            43.12
        ]
    },
    "properties": {
        "@id": "https://api.weather.gov/stations/KROC/observations/2025-08-07T02:40:00+00:00",
        "@type": "wx:ObservationStation",
        "elevation": {
            "unitCode": "wmoUnit:m",
            "value": 169
        },
        "station": "https://api.weather.gov/stations/KROC",
        "stationId": "KROC",
        "stationName": "Rochester, Greater Rochester International Airport",
        "timestamp": "2025-08-07T02:40:00+00:00",
        "rawMessage": "",
        "textDescription": "Clear",
        "icon": "https://api.weather.gov/icons/land/night/skc?size=medium",
        "presentWeather": [],
        "temperature": {
            "unitCode": "wmoUnit:degC",
            "value": 22,
            "qualityControl": "V"
        },
        "dewpoint": {
            "unitCode": "wmoUnit:degC",
            "value": 18,
            "qualityControl": "V"
        },
        "windDirection": {
            "unitCode": "wmoUnit:degree_(angle)",
            "value": 160,
            "qualityControl": "V"
        },
        "windSpeed": {
            "unitCode": "wmoUnit:km_h-1",
            "value": 7.416,
            "qualityControl": "V"
        },
        "windGust": {
            "unitCode": "wmoUnit:km_h-1",
            "value": null,
            "qualityControl": "Z"
        },
        "barometricPressure": {
            "unitCode": "wmoUnit:Pa",
            "value": 102743.03,
            "qualityControl": "V"
        },
        "seaLevelPressure": {
            "unitCode": "wmoUnit:Pa",
            "value": null,
            "qualityControl": "Z"
        },
        "visibility": {
            "unitCode": "wmoUnit:m",
            "value": 16093.44,
            "qualityControl": "C"
        },
        "maxTemperatureLast24Hours": {
            "unitCode": "wmoUnit:degC",
            "value": null
        },
        "minTemperatureLast24Hours": {
            "unitCode": "wmoUnit:degC",
            "value": null
        },
        "precipitationLast3Hours": {
            "unitCode": "wmoUnit:mm",
            "value": null,
            "qualityControl": "Z"
        },
        "relativeHumidity": {
            "unitCode": "wmoUnit:percent",
            "value": 78.062863996432,
            "qualityControl": "V"
        },
        "windChill": {
            "unitCode": "wmoUnit:degC",
            "value": null,
            "qualityControl": "V"
        },
        "heatIndex": {
            "unitCode": "wmoUnit:degC",
            "value": 22.293863671017778,
            "qualityControl": "V"
        },
        "cloudLayers": [
            {
                "base": {
                    "unitCode": "wmoUnit:m",
                    "value": 3810
                },
                "amount": "CLR"
            }
        ]
    }
}
```
</details>

### What did you expect to happen?

In case of a missing key there should be some indication in the logs, or display an error on screen?

### What actually happened?

Console indicated missing object error at line 221 of weathergov.js

### Participation

- [x] I am willing to submit a pull request for this change.

davidrea, 2025-08-07:
Just noticed age and version number on #3124 - looks like this remains an issue in 2.32.0. 

It appears the `?` chaining of the assignment was to try to deal with this situation, but if the `precipitationLastHour` child object simply isn't there, then referencing `.value` on it results in an error.

Looks like instead of:

```
currentWeather.precipitationAmount = currentWeatherData.precipitationLastHour.value ? currentWeatherData.precipitationLastHour.value : currentWeatherData.precipitationLast3Hours.value;
```

It should be:

```
currentWeather.precipitationAmount = currentWeatherData.precipitationLastHour ? currentWeatherData.precipitationLastHour.value : currentWeatherData.precipitationLast3Hours.value;
```

closed 2025-08-07 by davidrea

richardkiene, 2025-08-16:
@rejas I manually edited my weathergov.js and can confirm that this fix works.

rejas, 2025-08-16:
Cool. Can @sdetweil @KristjanESPERANTO  or @khassel take a look at the PR too?

davidrea, 2025-08-19:
❤️❤️ open-source! Thanks @khassel @rejas!

KristjanESPERANTO, 2025-09-30:
Release 2.33.0 is out 🙂

PR #3859 (2025-08-16): Thoroughly check for precipitationAmount values in weathergov provider
Fixes #3856