MagicMirrorOrg/MagicMirror #3884: Default Newsfeed Module Doesn't Handle Authentication with NY Times

raspberrypi9, 2025-09-09:
### Environment

MagicMirror² version: 
Node version: 
npm version: 
Platform:

### Which start option are you using?

node --run start

### Are you using PM2?

No

### Module

newsfeed

### Have you tried disabling other modules?

- [ ] Yes
- [x] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

When I send the show full article notification to the default newsfeed module, it used to display a full screen article text, but now it displays a captcha.  I have a NY times account but I don't know how the newsfeed module can be configured with my credentials so it will log in and be able to show the full article text once again.

### What did you expect to happen?

see previous comment

### What actually happened?

see previous comment

### Participation

- [ ] I am willing to submit a pull request for this change.

Desel72, 2026-03-10:
Hi @khassel, is this issue still active? If so, I’d be happy to work on it. Could you assign it to me?

KristjanESPERANTO, 2026-09-14:
I checked in with the other maintainers: We're closing this because it would add a lot of complexity to the newsfeed module and make it harder to maintain.

closed 2026-09-14 by KristjanESPERANTO