MagicMirrorOrg/MagicMirror #3901: [Bug] Can't fetch facebook calendar

lucasbotka, 2025-09-30:
### Environment

MagicMirror² version:  2.32.0
Node version: v22.20.0
npm version: 11.6.1
Platform: Raspberry Pi 4 Model B Rev 1.1 (arm64), Debian GNU/Linux 12 (bookworm)

Error:

 [2025-09-30 10:37:07.080] [ERROR] Calendar Error. Could not fetch calendar:  https://www.facebook.com/events/ical/upcoming/?uid=******&key=****** Error: Bad Request
0|mm       |     at NodeHelper.checkFetchStatus (/home/mirror/MagicMirror/js/node_helper.js:102:9)
0|mm       |     at process.processTicksAndRejections (node:internal/process/task_queues:105:5)

### Which start option are you using?

node --run start

### Are you using PM2?

Yes

### Module

calendar

### Have you tried disabling other modules?

- [x] Yes
- [ ] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

<details>
<summary>Configuration</summary>

```
 module: "calendar",
                        header: "Facebook události",
                        position: "top_left",
                        disabled: false,
                        config: {
                                calendars: [
                                        {
                                                symbol: "calendar-check",
                                                url: "https://www.facebook.com/events/ical/upcoming/?uid=******&key=******"                       }

```
</details>

Steps to reproduce the issue:
When I configure url for facebook events calendar, I am constantly getting error:

 [2025-09-30 10:37:07.080] [ERROR] Calendar Error. Could not fetch calendar:  https://www.facebook.com/events/ical/upcoming/?uid=******&key=****** Error: Bad Request
0|mm       |     at NodeHelper.checkFetchStatus (/home/mirror/MagicMirror/js/node_helper.js:102:9)
0|mm       |     at process.processTicksAndRejections (node:internal/process/task_queues:105:5)

### What did you expect to happen?

I tried to open url in web browser to confirm if it works and it started to download the .ics file which was populated with events, so there is no issue with the url. I have also tried as work around to download .ics file with curl and configure path to this file as url, but this is not supported. I have also noticed that curl only works with parameter -L so it follows redirects.

### What actually happened?

 [2025-09-30 10:37:07.080] [ERROR] Calendar Error. Could not fetch calendar:  https://www.facebook.com/events/ical/upcoming/?uid=******&key=****** Error: Bad Request
0|mm       |     at NodeHelper.checkFetchStatus (/home/mirror/MagicMirror/js/node_helper.js:102:9)
0|mm       |     at process.processTicksAndRejections (node:internal/process/task_queues:105:5)

### Additional comments

Seems that calendar fetcher cant handle redirects of facebook calendar.

### Participation

- [ ] I am willing to submit a pull request for this change.

sdetweil, 2025-09-30:
I think this is another UserAgent problem. 
Will test later. Tomorrows release has a new config option for this value

sdetweil, 2025-09-30:
when I use curl to access my fb events calendar I get 
```text
curl -sL https://www.facebook.com/events/ical/upcoming/?uid=100001xxxx49700&key=xsfW9Vxxxx2j9rr
[1] 2992696
 Sorry, this URL is invalid or has expired.
```

sdetweil, 2025-09-30:
researching this, I don't see anyone successful outside of google or apple in using the 'add event' link since 2021 or 2022

sdetweil, 2025-10-24:
we use nodejs fetch, 
by default it does redirect follow, you have to manually disable it
```
Default Behavior (redirect: 'follow'): By default, fetch will automatically follow HTTP redirects (status codes 3xx).
```